PaulStoffregen/DmxSimple #2: Problem with the serial connection

psyqjo, 2016-12-20:
when i add the line `Serial.begin(9600);` thenn i get the following error message:

C:\Users\PsyQjo\AppData\Local\Temp\buildf1aec6b8591d6c931cadcc7ef6c48c78.tmp/core\core.a(HardwareSerial0.cpp.o): In function `__vector_18':

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial0.cpp:48: multiple definition of `__vector_18'

libraries\DmxSimple\DMXSerial.cpp.o:C:\Program Files (x86)\Arduino\libraries\DmxSimple/DMXSerial.cpp:407: first defined here

C:\Users\PsyQjo\AppData\Local\Temp\buildf1aec6b8591d6c931cadcc7ef6c48c78.tmp/core\core.a(HardwareSerial0.cpp.o): In function `__vector_18':

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial0.cpp:48: multiple definition of `__vector_19'

libraries\DmxSimple\DMXSerial.cpp.o:C:\Program Files (x86)\Arduino\libraries\DmxSimple/DMXSerial.cpp:296: first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1

PaulStoffregen, 2016-12-20:
Please give me the complete program to reproduce this problem, so I can copy and paste it into Arduino and click Verify to get the error here.  I also need to know which version of Arduino and which board to select.

psyqjo, 2016-12-20:
i fixed it, i had the older version of dmx simple. now it works fine, thank you.

closed 2016-12-20 by PaulStoffregen