PaulStoffregen/DmxSimple #3: Erratic behavior on Uno

haarts, 2017-02-21:
I've been trying to get DMX to work with the official Tinkerit shield and an Arduino Uno. This library gave me the most success, but I'm not quite there yet.

Using the 'FadeUp' example I get an erratically blinking LED and using something like this `DmxSimple.write(3, 0);` in which 3 is my target channel, yields a fully lit LED where I expected an off LED.

I have the feeling that something is only slightly off but I have scant idea where to start looking. Perhaps something like https://github.com/PaulStoffregen/DmxSimple/commit/30fb70530f8f3716e72a25a0bc176f1a8c8c3aad?

PaulStoffregen, 2017-02-22:
Very similar question asked just a couple days ago:

https://forum.pjrc.com/threads/42210-DMX-512-devices-won-t-work

haarts, 2017-02-27:
Ah, this is going to hurt I think. :) 
Where did you get the constants like `__AVR_ATmega168__`? Perhaps it is as easy as adding the one for the Uno.

PaulStoffregen, 2017-02-27:
Use the Arduino forum.  Github issues are not the place to get project help.

This library does work on Arduino Uno.  You have done something wrong.  It's not the library.

closed 2017-02-27 by PaulStoffregen