PaulStoffregen/DmxSimple #5: Errant data at end of DMX packet

dannym83, 2017-03-21:
Hi There,

It seems that if I set maxChannels to 10 for example, then after the 10 sets have been sent out there is some data following which does get interpreted as some arbitrary value. 

In other words, if I configure to send 10 channels I then will receive 11 channels with the 11th channel having some arbitrary value. Anyhow, it shouldn't be there at all. 

I haven't looked into it in much detail but just noticed this when doing a quick test. I used an Enttec MK2 to monitor my DMX line.

Cheers

PaulStoffregen, 2017-03-21:
Which board are you using?  Can you give me a complete program to copy into Arduino?  I have limited time to do anything with this library, so please help me a bit here to help you...

dannym83, 2017-03-21:
Hi There,

I am using an UNO for this, I just took the ‘FadeUp’ example that comes with the library, no changes made to the code other then playing around with the number of channels in the setup code.

Cheers

Danny

dannym83, 2017-03-21:
Just looked at it with my a scope and I didn’t find anything after between the last byte sent and the start of the break before the next packet, that all looks fine.

I did notice quite some big gaps between the data send , between the first 8 slots and the 9th slot I measured 751us of space, though this is averagely less then 76us per slot I wonder if this is really necessary.  Anyhow, I don’t think this is causing the issue here.

However, I did noticed when I did a quick compare to another DMX signal and noticed that the slot timing (data rate) is shifting a bit.. (~6%) This might actually caused what I reported earlier on. I do know that the Enttec MK2 doesn’t have this issue reading DMX when I send DMX using the USART no matter if I only send 10 channels, 100  or a full DMX frame. (Just to rule out the possibility that the MK2 might be faulty on this)

Hope this helps a bit.

Cheers

Danny

dannym83, 2017-03-21:
Sorry,

Just to be exact. The 6% difference in data rate was measured over 5 slots of data.