PaulStoffregen/DmxSimple #6: Unexpected / uncontrollable DMX output

ecoluke, 2017-04-14:
Hi,

Hardware: 
1. I bought this shield: https://www.tindie.com/products/Conceptinetics/25kv-isolated-dmx-512-shield-for-arduino-r2/?pt=full_prod_search
2. I am using an Arduino UNO.

Preamble:
I initially installed the conceptinetics DMX library. I was able to get the light to behave normally - channel 1 is master dimmer, 2=R, 3=G, 4=B.

The conceptinetics library uses serial to send DMX and I ultimately need to use serial to control the DMX as in your SerialToDMX example - which I am very excited about using - so I am trying to make this library work for that reason.

Issue:
I installed the DMXSimple library and have managed to get *some light* to come out of my light - but it is confusing and makes no sense to me and I wonder if you might have any idea what is happening?

This is the code:

`#include <DmxSimple.h>

void setup() {
  
  DmxSimple.usePin(2);

  DmxSimple.maxChannel(32);
  
  /* Set channel 1 on the light (total dimmer) to max (255) 
  ** so that values for channels R (2), G (3) and B (4) are visible.*/
  DmxSimple.write(1, 255); 
  
}

void loop() {
  int x = 1;
   for (int i = 0; 256 > i > 0; i = i + x){
      // Fade the green channel in and out
      DmxSimple.write(3, i);
      if (i >= 255) x = -1; 
      if (i <= 0) x = 1;  
   } 
}`

What should happen is a green fade in and out - which works in the conceptinetics library. 

What comes out is irregular red flashes.

Other weirdness:
If I change the line DmxSimple.write(3, i); to DmxSimple.write(2, i); I still get red flashes but they are spaced differently. Any other channel - nothing.

If I change the line DmxSimple.maxChannel(64); to 32 or 4 or any other value then the red flashes also change their spacing.

Do you have any ideas or suggestions about what may be amiss as I feel like I am really missing a big part of the picture? 

Thanks,

Luke

ecoluke, 2017-04-15:
Solution:

To anyone trying to get this to work:

I managed to get it to work on an modern UNO using this shield: https://www.tindie.com/products/Conceptinetics/25kv-isolated-dmx-512-shield-for-arduino-r2/?pt=full_prod_search

You need to add in the following code to the setup() function:

  pinMode(2, OUTPUT);
  digitalWrite(2, HIGH);
  DmxSimple.usePin(4);

Then you need to set the jumpers:

EN
DE
TX-io
RX-io

haarts, 2017-05-02:
Sound similar to https://github.com/PaulStoffregen/DmxSimple/issues/3

simon-lucas, 2018-10-02:
I believe @ecoluke 's code may not be correct – at least for the non-isolated version of the DMX Shield I am using. So, with the DMXSimple library, I made sure the pin being set for output in setup(), match the pin being used by the shield. 

```
#define DMXPIN   4

void setup() {
  pinMode(DMXPIN, OUTPUT);
  digitalWrite(DMXPIN, HIGH);
  DmxSimple.usePin(DMXPIN);
  DmxSimple.maxChannel(5); //my number of channels

}
```

wyethan, 2019-04-12:
> I believe @ecoluke 's code may not be correct – at least for the non-isolated version of the DMX Shield I am using. So, with the DMXSimple library, I made sure the pin being set for output in [...]

I tried this code, and it didn't work at first.
But when I tried to change the parameter of DmxSimple.maxChannel() to 512 or 256 or 128 according to this discussion (https://www.reddit.com/r/arduino/comments/3tmhge/tinkerkit_dmx_master_shield_not_working_help/), it did work.