PaulStoffregen/DmxSimple #9: Arduino Mega 2560 compatibility?

TRSATOC, 2017-09-13:
Hello, my hardware is:
Arduino Uno (Official made in italy)
Arduino Mega (Chinese)
for DMX: a GROVE DMX 512 (http://wiki.seeedstudio.com/wiki/Grove_-_DMX512)

If I run the fadeUp example on the Uno + Grove DMX 512, I've got no problem, all is working
If I run the same example on the Mega + Grove DM 512, I've got nothing in the output
I've plugged my oscilloscope at the Mega port 3 output and I've got nothing so the GROVE isn't the cause of the problem.

I've tried a different port without success
I also try the blink example in the port 3 with success so my port 3 is working on the Mega.

I've never had problem with this Mega so perhaps there is a problem with the lib.

TRSATOC, 2017-10-16:
For information, i've tried with a nano and all is working perfectly on it :/
So what's happen with the Mega? (I've tried again today without success)

PaulStoffregen, 2017-10-16:
I tried just now.  With Arduino Uno:

[image]

I get this result:

[image]

With Arduino Mega (also a clone):

[image]

I get this result:

[image]

Both are running the same FadeUp example.  The 2nd byte changes rapidly when I view on the scope.

I do not know why it isn't working for you.  Everything I can see here seems to be fine with Arduino Mega.

TRSATOC, 2017-10-16:
I'm so sorry, I was using the Old library with no Mega 2560 support.
I'm very sorry for your waste of time, all is OK now

closed 2017-10-16 by TRSATOC

TRSATOC, 2017-10-17:
Thanks  @PaulStoffregen 
All is working well in my final implementation, I just have to play now to create my scenarys with my different DMX stuff!

[image]
[image]