PaulStoffregen/DmxSimple #12: not compiling - VSCode - PlatformIO - ESP8266

kurtjcu, 2018-06-12:
### Description

Describe your problem.
unable to compile using VSCode, Platform.io, for a wemos D1 (esp8266 based).

".piolibdeps\DmxSimple_ID898\DmxSimple.cpp:7:20: fatal error: avr/io.h: No such file or directory"

### Steps To Reproduce Problem

added header file to a project after installing the library via pio lib install

### Hardware & Software

Board Wemos D1 Mini
Shields / modules used none
Arduino IDE version VSCode using platform.io
Teensyduino version (if using Teensy) //na
Version info & package name (from Tools > Boards > Board Manager)
Operating system & version Win10
Any other software or hardware?

### Arduino Sketch

```cpp
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

#include <DmxSimple.h>

```

### Errors or Incorrect Output

If you see any errors or incorrect output, please show it here.  Please use copy & paste to give an exact copy of the message.  Details matter, so please show (not merely describe) the actual message or error exactly as it appears.

PaulStoffregen, 2018-06-12:
This library does not support ESP chips.

If anyone from the ESP community is willing to add the code and submit a pull request, I would be happy to merge (of course, if it doesn't mess with the non-ESP code...)

PaulStoffregen, 2018-06-13:
That's not my doing.  Please report this to the PlatformIO folks.

closed 2018-06-13 by PaulStoffregen