PaulStoffregen/DmxSimple #14: ATTiny85 support

Greenscreener, 2019-11-24:
Hello, 

I know I'm ignoring the issue template, but I think it is appropriate for this case. 

Would it be possible to add support for the ATTiny85 or is it not capable enough for that? 

Thank you very much.

PaulStoffregen, 2019-11-24:
This depends on what you mean by "possible".  If possible means someone with enough time & skill could write and test the code, then I'd say "probably".  I personally have not used ATTiny85, but I believe it has similar enough I/O.

But if "possible" means talking me into spending my engineering time on this, then the answer would be a hard no.

Greenscreener, 2019-11-24:
Ok then. I definitely do not have that skill. Thank you very much for your quick answer.

closed 2019-11-24 by Greenscreener