PaulStoffregen/DmxSimple #17: Not working on NodeMcu 1.0 (ESP-12E Module)

AndrazSpan, 2020-02-18:
I had tried to upload example sketch "FadeUp" to NodeMcu 1.0 (ESP-12E Module) but I get the following error message:

Arduino:1.8.11 (Windows 10), Plošča:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Users\andra\Documents\Arduino\libraries\DmxSimple\DmxSimple.cpp:7:20: fatal error: avr/io.h: No such file or directory

 #include <avr/io.h>

                    ^

compilation terminated.

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

LordSexy, 2021-03-30:
Hey. I had the same problem. Try the [ESP-DMX](https://github.com/Rickgg/ESP-Dmx) lib.
Works like a charm.
Be carefull what GPIO2 is. On my board its D4... Google is your best friend for this question

LordSexy, 2021-03-30:
https://github.com/LordSexy/esp-dmx-max485 Here are a few more examples.