PaulStoffregen/DmxSimple #21: Not working on Teensy 4

JonasVorwerk, 2020-10-20:
Can't get DmxSimple working on Teensy 4. 

I'm using the latest version of Teensyduino, DmxSimple, using the fade.ino example. 
With the same setup all dmx lights are fading using a Arduino Uno or a Teensy 3.2.

On Teensy 4 the code is compiling and uploading but all lights stay black.

andreatrona, 2021-07-24:
Hi Jonas, 
here working on a project and we had the same problem on teensy 4.1 . 
I think there are issues about 3.3V only of teensy 4.0 and 4.1 and some types on the library. Now I'm using a teensy 3.5 and work good for me.

I hope someone look at the source code and fix it, I'm not sure I'm able to do it...
bye

technodruide, 2022-03-22:
Hi there, i also use the pin 38 on a teensy 4.1, and i can confirm that it is not working. Nothing on the output (i used an oscilloscope). I used all the serials port for something else, so i really need to use a digital pin. I will try to add more details (eventually propose a correction if i understand the problem).

dimitre, 2022-07-13:
Does it work with some specific pin? Thanks

technodruide, 2022-07-13:
> Does it work with some specific pin? Thanks

I moved to another lib. Teensy dmx that use a Serial port.

dimitre, 2022-07-13:
@PaulStoffregen have you seen this issue? I've moved back to Teensy 3.2 today because it works. cheers

PaulStoffregen, 2022-07-13:
@dimitre - Yes, I am aware.  DmxSimple is a low priority because other DMX libraries work on Teensy 4.0 & 4.1, with better performance than DmxSimple.

fcana, 2023-07-24:
> @dimitre - Yes, I am aware. DmxSimple is a low priority because other DMX libraries work on Teensy 4.0 & 4.1, with better performance than DmxSimple.

Hi @PaulStoffregen , do you have some different libraries to advice for working with Teensy 4 ?

PaulStoffregen, 2023-07-24:
https://github.com/ssilverman/TeensyDMX